microsoft/vscode-arduino #1578: "arduino.enableUSBDetection": true causes Copilot failure to load

tpak, 2023-01-25:
Hello, 

Unbeleive able as the title may seem, setting  "arduino.enableUSBDetection": true will cause the Copilot extension to throw errors any time VSCode or a project is opened (Shift CMD-N). (Mac, M1, Ventura 13.1 all s/w is up to date - vscode version details at the end of the issue). 

Copilot will throw the following message:

[image]

output log:
```
[ERROR] [default] [2023-01-25T21:59:46.291Z] GitHub Copilot could not connect to server. Extension activation failed: "Cannot find module 'file-uri-to-path'
Require stack:
- /Users/chris/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.13/out/node_modules/bindings/bindings.js
- /Users/chris/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.13/out/node_modules/usb-detection/index.js
- /Users/chris/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.13/out/src/serialmonitor/usbDetector.js
- /Users/chris/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.13/out/src/arduino/arduino.js
- /Users/chris/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.13/out/src/extension.js
- /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js
- /Applications/Visual Studio Code.app/Contents/Resources/app/out/bootstrap-amd.js
- /Applications/Visual Studio Code.app/Contents/Resources/app/out/bootstrap-fork.js"
```

Setting  "arduino.enableUSBDetection": false will stop the error.

I have also reported this to GitHub support since it is coming from Copilot, but it seems deeply related to this extension and a singular setting, so I am bringing it over here. 

Reproducible steps (1... 2... 3...) that cause the issue
- install GitHub copilot
- install this Arduino extension
- enable Arduino extension setting for USB detection in the user preferences panel, is puts this in settings.json: "arduino.enableUSBDetection": true 
- restart VS Code 
- the error shown at the top occurs
- change the setting to false and re-launch vs code and the error stops

For deeper troubleshooting purposes, I did a "hard reset/reinstall" of VSCode as follows:

- I completely uninstalled all my extensions
- removed all user data stored in %APPDATA%\Code\User\settings.json
- I turned off settings sync and cleared my data from GitHub/VSCode's cloud
- I uninstalled and re-installed vscode
- I re-installed all my extensions (it threw a few meaningless errors since some were part of packages, but they all re-installed)
- I brought back co-pilot settings from settings.json
- I brought back Arduino settings
- the message comes back at this point
- so I narrowed it down, it is one setting in the Arduino extension that blows it all up
- "arduino.enableUSBDetection": true will cause this error with just the simple settings.json file below
- "arduino.enableUSBDetection": false and the error will go away

Removing all extensions except co-pilot and this Arduino extension, I can re-produce this each time.

install only two extensions:
code --install-extension vsciot-vscode.vscode-arduino
code --install-extension GitHub.copilot

super simple user preferences json:
```
{
    "arduino.enableUSBDetection": true,
    "editor.inlineSuggest.enabled": true
}
```

My code version info
```
Version: 1.74.3
Commit: 97dec172d3256f8ca4bfb2143f3f76b503ca0534
Date: 2023-01-09T17:05:47.808Z (2 wks ago)
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Darwin arm64 22.2.0
Sandboxed: No
```

tpak, 2023-01-25:
So, all in all, I can't be sure if this issue is resting on this extension or on Copilot, but I think I have narrowed the scope and it sort of points over here. If support comes back and accepts it as a

gcampbell-msft, 2023-02-21:
@tpak Thanks for your feedback. 

I would suggest you to keep the `enableUSBDetection` to false as a current solution/workaround. That support is currently broken anyway, so it would be a safe fix for the time being to keep that false. We will work to keep you updated on any future updates to the USBDetection. 

Thanks!

tpak, 2023-02-21:
@gcampbell-msft, that is what I have done, I didn't realize it was broken, but of course, I only develop for Arduino occasionally. I'll keep an eye out here for updates.