microsoft/vscode-arduino #1579: Select serial port name displays an imporper encoding.

cnKyana, 2023-01-28:
VScode displays a random number when i select serial port using this extension.
[image]
I have changed my arduino from 1.8.14 to 1.8.18 and reboot my PC but nothing happens.
VERSIONS
vscode: 1.74.3
os: Windows_NT x64 10.0.19045
extension: 0.4.13
arduino 1.8.18

gcampbell-msft, 2023-02-16:
@cnKyana We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue or if the issue still persists!

cnKyana, 2023-02-25:
Thanks for reply. I have tried new serial monitor, it's good but i encountered a new problem. The output info reveals an improper encoding when i downloading packages using the extension. The output looks like this, and i have no idea what causes this.
[image]
By the way, I am now in the university without my desktop which having the bug above. I tried to select a serial port using this laptop but failed with a "No serial port available" notice.

vscode: 1.75.1
extension: 0.5.0
Windows_NT x64 10.0.19045
Lenovo Laptop

gcampbell-msft, 2023-02-27:
@cnKyana Thanks for trying out the new experience!

A couple of clarifying questions:
1. Is this improper encoding problem happening in the Serial Monitor output? Or is it happening in one of the output panes? This will help us determine where the issue is happening. If you could provide a wider screenshot this also might help us know the answer to this question.
2. Do you have any ports connected to your laptop? If not, then this is expected. 
3. Are you remoting into your desktop from your laptop? If so, you may have to modify the `remote.extensionKind` setting for `ms-vscode.vscode-serial-monitor` in your `settings.json`. 

Thank you and I look forward to helping you diagnose this.

closed 2023-04-05 by github-actions[bot]