microsoft/vscode-arduino #1590: Arduino IDE is being passed a bad option at build time

ss32, 2023-02-13:
A bad (non-existing) argument is being passed to the arduino binary when trying to compile.  I cannot find a single reference to this failure anywhere else online.  

Arduino IDE 2.0.4-nightly-20230205

The code builds in the Arduino IDE.

```
/tmp/.mount_arduin87Xwqz/arduino-ide: bad option: --board
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch 'OLED_LoRa_Receive/OLED_LoRa_Receive.ino': Exit with code=9
```

The only reference I can find is on [arduino.ts:538](https://github.com/microsoft/vscode-arduino/blob/8555430d41a502ba7c81c436740bae5b7ad8f8b9/src/arduino/arduino.ts#L538), which is the `this.useArduinoCli()` block, and I'm **not** using the CLI per my settings; VS Code *should* be using the IDE.

[image]

The `arduino` binary that gets called is `arduino-ide_nightly-20230205_Linux_64bit.AppImage`

ss32, 2023-02-13:
For completeness I checked the latest stable, 2.0.3 and `--board` is missing as an argument there as well

```
Arduino IDE 2.0.3
Starting backend process. PID: 971301
Using browser-only version of superagent in non-browser environment
arduino-ide [workspace-directories] [options]

Options:
  --version                          Show version number               [boolean]
  --port, -p                         The port the backend server listens on.
                                                           [number] [default: 0]
  --hostname, -h                     The allowed hostname for connections.
                                                 [string] [default: "localhost"]
  --ssl                              Use SSL (HTTPS), cert and certkey must also
                                     be set           [boolean] [default: false]
  --cert                             Path to SSL certificate.           [string]
  --certkey                          Path to SSL certificate key.       [string]
  --app-project-path                 Sets the application project directory
                   [default: "/tmp/arduino-ide_2.0.3_Linux_64bit/resources/app"]
  --proxy-url                        Sets the proxy URL for outgoing requests.
                                                                        [string]
  --proxy-authorization              Sets the proxy authorization header for
                                     outgoing requests.                 [string]
  --strict-ssl                       Determines whether SSL is strictly set for
                                     outgoing requests.                [boolean]
  --log-level                        Sets the default log level
                   [choices: "fatal", "error", "warn", "info", "debug", "trace"]
  --log-config                       Path to the JSON file specifying the
                                     configuration of various loggers   [string]
  --root-dir                         DEPRECATED: Sets the workspace directory.
  --plugins                          Provides further refinement for the
                                     plugins. Example:
                                     --plugins=local-dir:path/to/your/plugins
                                                                        [string]
  --plugin-max-session-logs-folders  The maximum number of plugin logs sessions
                                     folders to retain. Example:
                                     --plugin-max-session-logs-folders=5
                                                          [number] [default: 10]
  --uncompressed-plugins-in-place    Whether user plugins that are stored on
                                     disk as uncompressed directories should be
                                     run in place or copied to temporary folder.
                                                      [boolean] [default: false]
  --extensionTestsPath                                                  [string]
  --pluginHostTerminateTimeout       Timeout in milliseconds to wait for the
                                     plugin host process to terminate before
                                     killing it. Use 0 for no timeout.
                                                       [number] [default: 10000]
  --pluginHostStopTimeout            Timeout in milliseconds to wait for the
                                     plugin host process to stop internal
                                     services. Use 0 for no timeout.
                                                        [number] [default: 4000]
  --vscode-api-version               Overrides the version returned by VSCode
                                     API 'vscode.version'. Example:
                                     --vscode-api-version=<Wanted Version>.
                                     Default [1.53.2]                   [string]
  --help                             Show help                         [boolean]
```

ss32, 2023-02-14:
Confirmed that pointing `arduino.path` to the older bash script that calls the Java build works without errors.

benmcmorran, 2023-02-16:
Thanks for the feedback. As you discovered, [only the legacy IDE is currently supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

closed 2023-02-16 by benmcmorran