microsoft/vscode-arduino #1595: Can't find library when switched to CLI

av4625, 2023-02-17:
I got a message in VS Code to say that the legacy IDE was going to be removed and to move to the bundled CLI. So I did. Now when I click the little verify button to build my project it can't find one of the libraries. The library is in my sketchbook `libraries` directory. 

It worked before switching to CLI and when I open the Arduino IDE 1.8.19 and click verify it works fine.

`fatal error: ezButton.h: No such file or directory`

```
% pwd
/Users/<user>/Documents/Development/Arduino/libraries
% ls ezButton 
README.md		keywords.txt		src
examples		library.properties
% ls ezButton/src 
ezButton.cpp	ezButton.h
```

I use a Mac.
Extension version v0.5.0

robotdad, 2023-02-18:
I do not believe the CLI shares the directories with the legacy IDE. Can you try restoring the libraries using the menus in VS Code?

MatthewSteeves, 2023-02-18:
I had same problem after accepting same prompt to swtich to CLI, and can confirm that @robotdad's suggestion worked for me! Thanks @robotdad!

VS code Output window showed a bit of encouraging but confusing output (note: I had extension's verbose logging turned on)
```
[Starting] Install library - Servo
Downloading Servo@1.1.8...
Servo@1.1.8 Servo@1.1.8 already downloaded
Installing Servo@1.1.8...
Installed Servo@1.1.8
[Done] Installed library - Servo
```

Manually running the `arduino-cli lib list` before and after:
```
PS C:\Users\Matt\.vscode\extensions\vsciot-vscode.vscode-arduino-0.5.0-win32-x64\assets\platform\win32-x64\arduino-cli> .\arduino-cli.exe lib list
No libraries installed.
PS C:\Users\Matt\.vscode\extensions\vsciot-vscode.vscode-arduino-0.5.0-win32-x64\assets\platform\win32-x64\arduino-cli> .\arduino-cli.exe lib list
Name  Installed     Available         Location              Description
Servo 1.1.8         -                 LIBRARY_LOCATION_USER -
```

I'll admit I find library management within the VS code extension a bit mysterious. For instance, now that I switched to the CLI, I'm not sure where the extension searches for libraries and where additional libraries paths can be configured. **Should I do it with the arduino-cli.exe directly?**

I've just seen that within the Arduino library manager in VScode, there's an "Installed" Type filter, and for each of the library's the filter shows, there's an with an option to "include library". That's great! While I was aware of the the Library Manager and Type filter, I hadn't considered that "Installed" would be a Type and so I didn't even bother to explore the drop-down menu closer. 🤦‍♂️ 

Now to just understand where the library path's are specified and config'd...

av4625, 2023-02-18:
I didn’t know you could install libraries through VS code, I’ll have a look to see how to do this.

Does anyone know where the libraries get stored when doing it this way? Specifically on Mac?

I had to fix/modify a couple of libraries to build for ESP32 as its compiler is more strict on warnings. For these I forked the library, added a fix commit and then installed the library manually by cloning my forked repo.

av4625, 2023-02-18:
```
[Starting] Remove library - ezButton
[Done] Removed library - ezButton

[Starting] Install library - ezButton
Downloading ezButton@1.0.4...
ezButton@1.0.4 ezButton@1.0.4 already downloaded
Installing ezButton@1.0.4...
Installed ezButton@1.0.4
[Done] Installed library - ezButton
```

Library manager in vscode showed it was already installed, I had to remove then reinstall. Doing a find on my system for the header of the library still shows it in the original location. Having to do this means I can't add libraries manually, is there away around this?

av4625, 2023-02-18:
I think this setting needs to be ported to the extension as I don't want to be forced to use `Documents/Arduino/libraries` it seems that the examples in the comment above is already using this setting to find the libraries.
[image]

av4625, 2023-02-18:
Would be nice if [--export-binaries](https://arduino.github.io/arduino-cli/0.30/commands/arduino-cli_compile/#options) could be used as well

rtnate, 2023-03-01:
I fixed this by setting an environment variable ARDUINO_DIRECTORIES_USER to match my sketchbook directory on an extra hard drive (in my case on windows, E:\source\arduino).  You will have to restart vscode for it to pull it in.  I did it quickly through System Properties.

cechurat, 2023-03-18:
Following works for me on Windows:
PowerShell: `cd .\.vscode\extensions\vsciot-vscode.vscode-arduino-0.6.0-win32-x64\assets\platform\win32-x64\arduino-cli\`
`.\arduino-cli.exe config init`
`.\arduino-cli.exe config set directories.user "<YOUR_SKETCHBOOK_LOCATION>"`

If you want to check it type:
`.\arduino-cli.exe config dump` and you should see the path you entered before.

Toggle the VS Code and try to Verify your project. In file .vscode\c_cpp_properties.json check the property "includePath" which should reflect your changes.

vmajor, 2023-12-19:
> I do not believe the CLI shares the directories with the legacy IDE. Can you try restoring the libraries using the menus in VS Code?

What do you mean? I too am horribly stuck trying to migrate to VSC from Arduino IDE. VSC with the bundled arduino-cli cannot find my libraries and I have no idea how to help it.

gcstr, 2024-08-17:
I have the same problem here. The libraries are installed, but the VSCode plugin won't recognize it.

[image]

My CLI reports that the libraries are installed just as VSCode:

```
➜ arduino-cli lib list
Name            Installed Available     Location Description
FTDebouncer     1.3.3     -             user     -
lvgl            8.3.10    9.1.0         user     Full-featured Graphics Library for Em...
TFT_eSPI        2.5.34    2.5.43        user     TFT graphics library for Arduino proc...
TFT_eSPI_Setups           -             user     -
```