microsoft/vscode-arduino #1597: Unable to locate library file when compiling with VSCode but fine with ArduinoIDE

reflectingme, 2023-02-18:
I appear to have an issue with libraries when trying to compile and or upload with VSCode.

When I use either of the Arduino IDE's to compile and upload the code, it works without any issues and the app runs fine on my ESP32S3 board.  However, when I do the same thing in VSCode, I get the error unable to locate the file "TFT_eSPI.h" as indicated by the error underline in screenshot. 

I have of course ensured the TFT_eSPI library is installed correctly, which it is as the sketch runs fine when compiled with Arduino IDE, but just not with VSCode.

Could someone please help and point me in the right direction please.  I am not fond of the Arduino IDE and would much rather use VSCode or even VS2022 which I use for playing around with c# and .net.

Many thanks.

[image]

benmcmorran, 2023-02-22:
Can you try installing the library that provides `TFT_eSPI.h` using the library manager in VS Code? Run the "Arduino: Library Manager" command in the VS Code command palette. If you started using the Arduino CLI in the latest version of this extension, libraries may be in a different directory now (see #1595).

reflectingme, 2023-03-15:
Thanks Ben... I have managed (I think) to figure out what was happening.  It really is to do with my lack of experience and knowledge on #include paths.  I updated the include path in the configuration within VSCode, and have _sort of_ got it to work.  I need to research/kearn what the /** means at the end of each include line, and how to ensure I add the full relative path to any files I am working with.

Sorry to have used your time here for what must be, for you, a trivial issue.

closed 2023-04-21 by github-actions[bot]