microsoft/vscode-arduino #1598: Not Analyzing the Code

jeevanghimire, 2023-02-19:
Analyzing sketch 'BlindStick\blindStick.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
Error opening sketch: no valid sketch found in C:\Users\Administrator\Desktop\Jeevan\Rust\BlindStick: missing C:\Users\Administrator\Desktop\Jeevan\Rust\BlindStick\BlindStick.ino
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch 'BlindStick\blindStick.ino': Exit with code=1

My code 

// make a program that sense the envroment using ultrasocnic sensor and play the buzzer

// define the pins
const int trigPin = 9;
const int echoPin = 10;
const int buzzerPin = 7;

// define the variables
long duration;
int distance;

void setup()
{
    // put your setup code here, to run once:
    Serial.begin(9600);
    pinMode(trigPin, OUTPUT);
    pinMode(echoPin, INPUT);
    pinMode(buzzerPin, OUTPUT);
}

void loop()
{
    // put your main code here, to run repeatedly:
    digitalWrite(trigPin, LOW);
    delayMicroseconds(2);
    digitalWrite(trigPin, HIGH);
    delayMicroseconds(10);
    digitalWrite(trigPin, LOW);
    duration = pulseIn(echoPin, HIGH);
    distance = duration * 0.034 / 2;
    Serial.print(distance);
    Serial.println("cm");
    if (distance > 50 && distance <100 )
    {
        tone(buzzerPin, 500);
    } 
    if else (distance <50){
      tone (buzzerPin,1000);
    }
    else
    {
        noTone(buzzerPin);
    }
}

// make a program that sense the envroment using ultrasocnic sensor and play the buzzer

gcampbell-msft, 2023-02-21:
Hi, @jeevanghimire!

Thanks for bringing this to our attention. I believe that the name of your sketch file must _exactly_ match that of the folder that it's in. Could you try ensuring that the casing of your file is the same as the folder and try to verify again?

krid78, 2023-02-28:
This fixed a similar issue I had.
Will this change in the future?

gcampbell-msft, 2023-02-28:
@krid78 It's hard to say. This is a design choice by Arduino.

Thanks for all of your feedback!

krid78, 2023-02-28:
Thanks for the fast reply. I was not aware of this requirement by the Arduino IDE.

Until they change this, the solution you proposed is correct.

To summarize my situation, here is the error I got.

```none
[Starting] Analyzing sketch 'code\arduino_uno_led_001.ino'
Please see the build logs in output path: <basepath>\build
Error opening sketch: main file missing from sketch: <basepath>\code\code.ino
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch 'code\arduino_uno_led_001.ino': Exit with code=1
```
Chaning the name of the Sketch fixed the issue.

While analyzing the issue, I defined a `"output": ...` in the `arduino.json` to get rid of the message:

```
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
```

Now everything works as expected.

julesgilson, 2023-03-16:
This did not use to be the case. I rarely use Arduino, but just opened an existing package (that was working fine) and now it is requiring a sketch file to be named the same as the folder... Traditionally I have always prefixed sketch files with a number - to ensure that includes were included properly (used to be an issue - don't know if it still is) and defined the primary sketch in the arduino.json file.

closed 2023-04-22 by github-actions[bot]