microsoft/vscode-arduino #1599: EPERM when attempting any operation

flipphillips, 2023-02-20:
Using latest build (0.5.0) on M1 macOS.

## to replicate
Attempt to verify code, lode library manager, etc, using embeded CLI 

## expected result
Activity executed

## actual result
```
A system error occurred (EPERM: operation not permitted, chmod '/Users/flip/.vscode/extensions/vsciot-vscode.vscode-arduino-0.5.0-darwin-arm64/assets/platform/darwin-arm64/arduino-cli/arduino-cli.app')
```

## regression
Compiles fine through Arduino IDE. 
`arduino-cli.app` is already executable?

benmcmorran, 2023-02-21:
Thanks for the report! I'll start working on a fix for this issue, which looks related to #1584, but in the meantime as a workaround you should be able to manually [download Arduino CLI from GitHub](https://github.com/arduino/arduino-cli/releases/tag/0.30.0), then set `arduino.path` and `arduino.commandPath` to point at the downloaded Arduino CLI and ensure `arduino.useArduinoCli` is `true`.

benmcmorran, 2023-02-22:
@flipphillips I don't currently have M1 hardware to test on, but I wasn't able to reproduce this issue on a x64-based Mac.

- Are you running VS Code under a different user account that you originally used to install the extension?
- Can you check which user is considered the owner of the `/Users/flip/.vscode/extensions/vsciot-vscode.vscode-arduino-0.5.0-darwin-arm64/assets/platform/darwin-arm64/arduino-cli/arduino-cli.app` file, and what its permissions currently are?

The extension needs to be able to change file permissions on that file at least once because it's not executable by default, but I can add extra logic to not attempt to change executable permissions again if they are already set.

cliffxzx, 2023-02-22:
same issue

flipphillips, 2023-02-22:
> Thanks for the report! I'll start working on a fix for this issue, which looks related to #1584, but in the meantime as a workaround you should be able to manually [download Arduino CLI from [...]

In the past, I had downloaded the cli via `brew`. I wasn't able to get it to work correctly with the plugin, regardless of the path settings in VS Code.

```
$ arduino-cli config dump
board_manager:
  additional_urls: []
daemon:
  port: "50051"
directories:
  data: /Users/flip/Library/Arduino15
  downloads: /Users/flip/Library/Arduino15/staging
  user: /Users/flip/Documents/Arduino
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
output:
  no_color: false
sketch:
  always_export_binaries: false
updater:
  enable_notification: true
```

as for the app as installed in the plugin directory, it is very much owned by me, and was executable from the git-go

```
-rwxr-xr-x@ 1 flip  staff    27M Feb 20 13:41 /Users/flip/.vscode/extensions/vsciot-vscode.vscode-arduino-0.5.0-darwin-arm64/assets/platform/darwin-arm64/arduino-cli/arduino-cli.app*
```

benmcmorran, 2023-02-22:
@flipphillips @cliffxzx I have a new build with the fixes from #1601 ready for testing. If you have time, please download [this build artifact](https://github.com/microsoft/vscode-arduino/suites/11140220102/artifacts/568222070), extract it, and then from within VS Code run the "Install from VSIX..." command and select the appropriate platform-specific VSIX file. For M1 Mac, that would be `vscode-arduino-darwin-arm64.vsix`. The changes in this build will skip attempting to `chmod` if the executable bit is already set and will provide a better error message if there was a problem setting the executable bit. Let me know if you see any issues and thanks again for your feedback!

flipphillips, 2023-02-22:
Absolutely will in a little bit, just in time to show students how to use it :)

—
flip phillips, phd
professor / motion picture science + imaging science / rit
flipphillips.com

benmcmorran, 2023-02-22:
Great, thanks for confirming. I'll try to get this fix out soon. IntelliSense only updates after a successful compile, but there may be other things going on that are impacting IntelliSense. We're also planning to overhaul how we get IntelliSense data in a future release.

flipphillips, 2023-02-22:
The preferences are a little wonky since it tries to assemble a path from the pieces if I hard-code them, but on the Mac version the cli lives in a different place than the rest of the cli-tools, so that may or may not be an issue also.

jseyster, 2023-03-11:
It's possible that the chmod operation is getting blocked by Gatekeeper. I encountered this error on my M1 Mac as well, and I was able to work around it by going to "Privacy & Security" in System Settings, navigating to "App Management," and enabling access for "Visual Studio Code." After restarting Visual Studio Code, this error stopped appearing.

There is the downside that adding this permission disables some protection from malicious extensions, though, so skipping the chmod when arduino-cli already has executable permissions could still be beneficial. Hope that helps!

fredsa, 2023-11-05:
In case its helpful for this issue, I found an `arduino-cli.app` file attribute (permission) issue after upgrading an M1 Macbook Pro to Sonoma. I had a working Arduino extension that stopped working when I upgraded to Sonoma.

See https://github.com/microsoft/vscode-arduino/issues/1681#issuecomment-1793850776 for details, including before/after file permission bits.