microsoft/vscode-arduino #1600: Writing to Lilygo T-display-S3 after some % of the board is loaded

Ferrari2, 2023-02-20:
Hi
When I am using Visual Studio Code to program my Lilygo T-Display-S3, the writing stops after some lines showing "Writing at..." (see "Programming sequence" below). Sometimes the programming is working fine.

I have tried changing the serial programming speed from 15200 to 19200 and still see the same issue.

Programming another sketch via the Arduino IDE is working as a charm on the same T-Display-S3.

Any suggestions to solve the issue.

**Programming sequence:**
...
Looking for upload port...
Auto-detected: COM4
Uploading .pio\build\ESP32-S3-DevKitC-1\firmware.bin
esptool.py v4.4
Serial port COM4
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:22:c2:b0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000e8fff...
Flash params set to 0x023f
Compressed 15008 bytes to 10328...
Writing at 0x00000000... (100 %)
Wrote 15008 bytes (10328 compressed) at 0x00000000 in 0.3 seconds (effective 398.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 271.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 448.7 kbit/s)...
Hash of data verified.
Compressed 887856 bytes to 572054...
**Writing at 0x00010000... (2 %)
Writing at 0x0001a4e3... (5 %)
Writing at 0x00027089... (8 %)**
Traceback (most recent call last):
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool.py", line 34, in <module>
    esptool._main()
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 1026, in _main      
    main()
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 826, in main        
    operation_func(esp, args)
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool\cmds.py", line 556, in write_flash     
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 109, in inner
    return func(*args, **kwargs)
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 991, in flash_defl_block
    self.check_command(
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 407, in check_command 
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 376, in command       
    p = self.read()
  File "C:\Users\oz2bk\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration
*** [upload] Error 1
====================================== [FAILED] Took 13.63 seconds ======================================

gcampbell-msft, 2023-02-21:
@Ferrari2 Have you tried a baud rate of 9600? I did a quick search and that may be the required baud rate. 

Let us know if this fixes your issue, and thanks for bringing it to our attention!

Ferrari2, 2023-02-21:
Hi gcampbell-msft
It unfortunately seems to have no effect using 9600 kbit/s.
Do you know why the serial speed changes like this (from the log above):
**Changing baud rate to 460800
Changed.**
Could it be the issue and where can I change it to a lower speed?

closed 2023-03-31 by github-actions[bot]