microsoft/vscode-arduino #1602: How to tell vscode to check for libraries folder ?

eagl1, 2023-02-22:
Hi,

I'm trying to compile my ESP32 code, but vscode don't know where my "libraries" folder is.

[https://europe1.discourse-cdn.com/arduino/original/4X/4/c/b/4cbe09aae20fa02e97a6956bb91f5c98d39dbcce.png](url)

How to set the folder's path ?

Almost same problem as #1033

benmcmorran, 2023-02-22:
Can you try installing the library that provides `TFT_eSPI.h` (and any other headers you need) using the library manager in VS Code? Run the "Arduino: Library Manager" command in the VS Code command palette. If you started using the Arduino CLI in the latest version of this extension, libraries may be in a different directory now (see #1595).

eagl1, 2023-02-22:
OK, after a many error/trail, surprisingly the error went away :)

[image]

eagl1, 2023-02-23:
I solved part of the problem with these settings:

`            "includePath": [
                "${workspaceFolder}/**",
                "D:/Program_Files/Arduino/hardware/arduino-esp32-master/libraries",
                "D:/Program_Files/Arduino/hardware/arduino-esp32-master/cores/esp32",
                "D:/Program_Files/Arduino/libraries",
                "D:/Program_Files/Arduino/hardware/tools/avr/lib/gcc/avr/7.3.0/include"
            ],`

But I moved the project folder and got the errors again !

Working with vscode isn't easy.

eagl1, 2023-02-26:
I think first I have to do the following:
1. Compile the code without the included libraries.
2. vscode will generate necessary ".json" files in the ".vscode" folder, for the compiled code. Like this:

[image]

3. After that, the programmer can add the libraries path in the "c_cpp_properties". Like this:

[image]

MickGyver, 2023-03-10:
It seems that on Windows, libraries must be in the C:\Users\[user]\Documents\Arduino\libraries folder, then they will be found by the compiler. I tried @eagl1 's trick of compiling without the headers first to get the c_cpp_properties.json file but the only way I'm able to compile is if I copy my libraries to the folder I mentioned above.

eagl1, 2023-03-10:
Yes, that is on Windows.

You can use the path `C:\Users[user]\Documents\Arduino\libraries`, and I'm sure this is for the new Arduino IDE 2.0, as I'm working with the legacy Arduino IDE. I'm happy I learned how to setup this wonderful IDE.

But I got back to the legacy Arduino IDE, it's more convenient for me more than vscode and the new Arduino IDE.

oskardev, 2023-03-12:
Maybe it will be helpful - maybe not - I had a project already created in the Arduino IDE then I decided to use the Arduino extension for VS Code.
I was able to verify the project and upload - no problems detected. But the headers were highlighted - like on the screenshots shown earlier. Of course it was not possible to jump to the header too. But the compiler somehow didn't have the problems.
In the c_cpp_properties.json I had all needed paths (I didn't add it manually).

So I created the new folder, new .ino file, just pasted the content from the existing project - it was needed only to set the port, board and the output folder - no build problems, no problems with uploading.

I checked the old and new cofig files in .vscode i neach project. Looks like we have some differences.

arduino.json:
[image]

c_cpp_properties.json (additional content in the old project):
[image]

c_cpp_properties.json again with this diff section and full includePath content:
[image]

So I copied the new config files to the old project - now looks like everything is ok.

eagl1, 2023-03-13:
1. The differences in arduino.json are just in the sequence of the arguments.
2. The workspaceFolder should contain your application libraries. Like this:

[image]

3. The includePath should contain the hardware libraries.