microsoft/vscode-arduino #1603: Cant upload arduino ino file

YuMERA, 2023-02-22:
Type: <b>Bug</b>

I cant uploas arduino ino file. I have this error message : Unhandled exception when cleaning up build "Uploading": {}

I also cannot change the selected serial port and the Select programmer is empty, I have no choice

Extension version: 0.5.0
VS Code version: Code 1.70.0 (da76f93349a72022ca4670c1b84860304616aaa2, 2022-08-04T04:37:57.358Z)
OS version: Windows_NT ia32 10.0.19045
Modes:

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-2310 CPU @ 2.90GHz (4 x 2893)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: disabled_off<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>opengl: enabled_on<br>rasterization: unavailable_off<br>raw_draw: disabled_off_ok<br>skia_renderer: enabled_on<br>video_decode: enabled<br>video_encode: unavailable_off<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|2.99GB (0.47GB free)|
|Process Argv|--crash-reporter-id 8803f4c3-b66b-4da5-ad40-fe50a64f71ee|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368cf:30146710
vsreu685:30147344
python383:30185418
vspor879:30202332
vspor708:30202333
vspor363:30204092
vslsvsres303:30308271
pythonvspyl392:30443607
vserr242:30382549
pythontb:30283811
vsjup518:30340749
pythonptprofiler:30281270
vshan820:30294714
vstes263cf:30335440
pythondataviewer:30285071
vscod805cf:30301675
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
cmake_vspar411:30581797
vsaa593cf:30376535
pythonvs932:30410667
cppdebug:30492333
vsclangdc:30486549
c4g48928:30535728
dsvsc012:30540252
pynewvextcfv2:30669238
azure-dev_surveyone:30548225
pyindex848:30662994
nodejswelcome1cf:30587006
3biah626:30602489
pyind779cf:30662993
89544117:30613380
pythonsymbol12:30657548
showlangstatbar:30664136
vscodedisable:30660115

```

</details>

RcrossF, 2023-02-23:
I have the same problem, the serial port can be set in `.vscode/arduino.json` but that doesn't fix the upload bug.

YuMERA, 2023-02-24:
After updating VSCode everything is fine again. Thx

closed 2023-02-24 by YuMERA

alexCajas, 2023-02-25:
The upgrade didn't work for me, some idea?

foxyisnotsalt, 2023-03-05:
same problem here, i'm using windows 7