microsoft/vscode-arduino #1615: I cannot upload a code in arduino, every time appears this error message: Unhandled exception when cleaning up build "Uploading": {}

foxyisnotsalt, 2023-03-05:


trethaller, 2023-03-07:
I also have this problem

gcampbell-msft, 2023-03-08:
@Pharap 
I'm curious about you not being able to manually select a port for weeks. When did this issue begin? We recently updated the extension such that it depends on the Serial Monitor extension, so I'd like to hear more about this issue, what the UI looks like when you attempt to select a port, what versions of the extension that it broke on, etc. When you try to manually select a port, is it failing? Or is it not showing any ports? 

Any additional information that you can provide would be great!

trethaller, 2023-03-08:
I also cannot select ports. Nothing happens (no error) when I click on the "select port" button.

Pharap, 2023-03-08:
@gcampbell-msft

>When you try to manually select a port, is it failing? Or is it not showing any ports?

Like @trethaller, I click the `<Select Serial Port>` button on the bottom right and nothing happens. No menu, no error message, nothing.

>When did this issue begin?

I'm not certain. Some time after the move to CLI. Possibly coincident with that.

>what versions of the extension that it broke on

I'm currently on Arduino v0.5.0 and Serial Monitor v0.7.0.

trethaller, 2023-03-08:
@gcampbell-msft thanks for catching the issue, that was indeed fixed by upgrading Serial Monitor to 0.8+, which version didn't appear because it depended on a newer VScode build and I had auto-updates disabled. 
Now I experience a different error when uploading through CLI (`unknown flag: --build-path`) but at least this one issue is now fixed. Thanks.

Pharap, 2023-03-08:
@gcampbell-msft

I actually primarily use [VSCodium](https://vscodium.com/) to avoid telemetry. The unfortunate trade-off is that it's unable to use Microsoft's extension network, it has to maintain [a separate network](https://open-vsx.org/) (to which only some extensions are published) so I tend to put off updating extensions until I really need to.

Hence I had to update/reinstall VSCode before I could update Serial Monitor.

It took quite a while. In addition to the seven extensions that I actually use, VSCode insisted on installing a load of extensions I didn't really want and then had to uninstall (e.g. Learn Markdown, Python, Pylance, Jupyter - I almost never use Python and already know Markdown). After three or four restarts I was finally up-to-date.

Then for good measure I updated VSCodium and copied across the extensions...

Only to discover that Serial Monitor thinks VSCodium is VSCode version 1.68 and says it won't work with anything lower than version 1.74.

I can at least confirm that it works on VSCode, just not yet on VSCodium.

Seems I'll just have to make do without it for now.

gcampbell-msft, 2023-03-09:
@Pharap Your most recent issue is posted here at #1646, so please reference that issue to track the progress.

For your issue related to VS Codium: VS Codium is not officially supported. 

Closing this bug in favor of #1646.

closed 2023-03-09 by gcampbell-msft