microsoft/vscode-arduino #1616: Cannot activate the Arduino extension

ae-warp, 2023-03-05:
vscode-arduino version: 0.5.0
vscode-serial-monitor version: 0.8.0
Visual studio code version: 1.76.0
Windows build: 19044.2604
Raspbian Desktop version: 11 (bullseye)
Raspberry Pi OS Lite version: 11 (bullseye)

**Reproducible steps:**
You'll need an Arduino (in my case a Due, but one can use Uno or any other), plugged in a machine to which you log into via e.g. vscode Remote SSH plugin.
VSCode's host OS is Win10, remote host is a Raspberry (the one to which Arduino is plugged in) with either RPi OS Lite or Raspbian Desktop OS version. 
Tests have been made for both vscode-arduino and vscode-serial-monitor with release and pre-release versions. 
RPi OS Lite has been installed with the purpose to test this issue.

**What you expected to see, versus what you actually saw:**
1. (with 0.4.13 plugin version) I was able to write, debug, compile and upload code on the Arduino board, open Serial Monitor, etc. using the configuration described above.
2. (with 0.5.0 plugin version) I can't use the same chain of tools to work with Arduino. I get this error "arduino.openSerialCommand not found", and the like. 

**Images, animations, or a link to a video showing the issue occurring**: 
nothing to say, normal and expected UI behaviour.

**A code snippet that demonstrates the issue**: 
none.

**Errors from the Dev Tools Console**:
This one (available on UI and Dev Tools>Console): "Cannot activate the 'Arduino' extension because it depends on the 'Serial Monitor' extension, which is not loaded. Would you like to reload the window to load the extension?" (Reloading does not solve the issue).
Serial Monitor tip for remote host plugin instance says: "This extension is enabled in the Local Extension Host because it prefers to run there."

Last consideration:
I think [Modify to use standalone Serial Monitor Extension](https://github.com/microsoft/vscode-arduino/pull/1577)  (#1577) is relevant.

Thank you for your time.

Alexia, 2023-03-07:
+1 from me.  This is broken to the point that I can not even uninstall the Serial Monitor extension in an attempt to fix this.  It claims that the Arduino extension requires it even after I uninstall the Arduino extension.

ae-warp, 2023-03-08:
> Can you try adding this snippet to your settings.json to force the Serial Monitor to run on the remote?

Everything below is under this configuration: vscode arduino 0.5.0 and vscode serial monitor 0.8.0 versions.

So I configured the json with the snippet you posted - much appreciated, really, and my bad, couldn't find the issue on that repo. Anyway, with that I no longer see the "Serial Monitor Extension not loaded" exception.

**PLEASE NOTE**: until I withdraw this below, in this scenario I have vsc arduino and vsc serial monitor installed locally and remotely.

I get this error, from the dev tools console

```
  [workbench.desktop.main.js:615 ]
  ERR Could not find pty on pty host: CodeExpectedError: Could not find pty on pty host
    at S.L (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:27:6849)
    at S.updateTitle (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:27:2630)
    at Object.call (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:17:5043)
    at l.s (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:15:5412)
    at l.q (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:15:4915)
    at /home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:15:4326
    at n.invoke (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:11:145)
    at E.deliver (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:11:2070)
    at c.fire (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:11:1708)
    at process.T (/home/pi/.vscode-server/bin/92da9481c0904c6adfe372c12da3b7748d74bdcb/out/vs/platform/terminal/node/ptyHostMain.js:9:20949)
    at process.emit (node:events:526:28)
    at emit (node:internal/child_process:938:14)
    at processTicksAndRejections (node:internal/process/task_queues:84:21)
```
Please note that the above error can be present more than once, even consecutively.
Finally, I get this in OUTPUT panel
```
[Starting] Analyzing sketch '/whatever/dir/script/script.ino'
Please see the build logs in output path: /whatever/dir/arduino/.build
Error: unknown option: --no-color
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch '/whatever/dir/script/script.ino': Exit with code=3
```
Tried to grep the ```--no-color``` option: honestly, I get a bunch of results from anywhere inside the project (i also use npm, dunno how much this weighs), that I don't know what to post. Nothing seemed relevant to me, though, or somehow related to my actions.

So what I did was uninstalling both extensions from local and remote envs and reloading vscode.
Subsequently I installed the extensions only in the remote host: I see, in Running Extensions, both of them working only in remote, as expected. 
I then open my script and after clicking on verify button I get this tip:
```Cannot find Arduino tools. Use Arduino CLI bundled with this extension instead?```
After selecting "Use bundled arduino-cli", vscode reloads. I then see the same loop of errors, warnings, etc.

How can I be of further help? 
I got very few time right now to further investigate this: suggestions on next steps would be most appreciated.

Thank you again for the snippet and, lbnl, for your time.

benmcmorran, 2023-03-08:
I'm guessing the `--no-color` flag is coming [from here](https://github.com/microsoft/vscode-arduino/blob/5f2cf62ad4eb1e5b29084d1b3324eb6cf9e52639/src/arduino/arduino.ts#L685), which was changed relatively recently to improve the IntelliSense configuration generation for Arduino CLI. I'm not sure why that flag wouldn't be available from Arduino CLI unless perhaps the version of the CLI was very old.

It's concerning that the bundled CLI doesn't seem to be initializing correctly. We can try to reproduce that case in a remote environment, but as a workaround can you try manually installing [Arduino CLI](https://github.com/arduino/arduino-cli/releases/tag/0.31.0) in your remote environment, then set `arduino.useArduinoCli` to `true`, `arduino.path` to the path to the directory containing the Arduino CLI executable, and `arduino.commandPath` to the name of the Arduino CLI executable (probably just `arduino-cli` on Linux). Reload VS Code and in theory it will correctly find the CLI in the remote environment.

arsegorov, 2023-03-09:
I get the same `--no-color` error, and I'm using the CLI packaged with the extension (Arduino extension 0.5.0).

ae-warp, 2023-03-10:
@benmcmorran 
Hello again; sorry for the delay, couldn't get back earlier to the project.
So, I followed your suggestion and installed arduino-cli.
Now the extension works as expected at least for what I tested - verify, upload, board manager, serial monitor. 

Summing it up: I can work with the most recent version of arduino extension for vscode, though I won't close the thread, leaving it to you to decide.
Thank you very much, I really appreciated your feedback and help.

PS: I have a spare SD that I can experiment with on RPi. Please let me know if I can be of any help.

closed 2023-04-16 by github-actions[bot]