microsoft/vscode-arduino #1619: Including libraries on Arduino CLI

VSR-Adithya, 2023-03-12:
I am currently using Arduino CLI version `0.31`. I have used Arduin IDE in the past. So, I got all my installed libraries into cli using `arduino-cli lib install`. I use Arduino CLI on visual studio code using Arduino extension by Microsoft. 
I Even though I have my libraries installed I was not able to use them on my code like I used to on IDE. For example `#include <Servo.h>`. Help me if I am missing something

JohnboyK, 2023-03-21:
My suggestion: On your search with file explorer in the directory where you have installed VsCode to a subdirectory similar this one: <install-directory of VsCode>\data\extensions\vsciot-vscode.vscode-arduino-0.6.0-win32-x64\assets\platform\win32-x64\arduino-cli
Now open a command window and change to this directory. When you enter the command "ardulino-cli lib list" your library is listed correctly with the location "LIBRARY_LOCATION_USER"?  Alternatively enter the command "arduino-cli config dump". Here you get in the section "directories" the entry "user: C:\Users\<name>\.......\Arduino".  Note this directory and look with file explorer if you can find there a subdirectory C:\Users\<name>\.......\Arduino\libraries. Is there the installed the library which you use?

rrehrig-SF, 2023-12-15:
Your question is a little vague, but it sounds like something I just solved for myself. For me, neither the Arduino extension for vscode nor the included arduino-cli were pulling in any of the built-in Arduino libraries (SD, servo, TFT, etc). I would get the build error `fatal error: SD.h: No such file or directory` when trying to compile using `#include <SD.h>` in my sketch.

To fix this, I used the arduino-cli to create a configuration file (`arduino-cli config init`). I then edited that config file (arduino-cli.yaml located in the Arduino15 directory) to add the `builtin.libraries` field under `directories` section with a path to the built-in libraries in the Arduino15 directory.

My directories section of arduino-cli.yaml roughly looks like the following:

```
directories:
  data: /path/to/Arduino15
  downloads: /path/to/Arduino15/staging
  user: /path/to/Documents/Arduino
  builtin.libraries: /path/to/Arduino15/libraries
```

More details on the arduino-cli configuration https://arduino.github.io/arduino-cli/0.35/configuration/.