microsoft/vscode-arduino #1621: How to see more output when Compiling: Verifying and Uploading

reflectingme, 2023-03-15:
Just curious about this, but when I use ArduinoIDE2 and click the Upload button, I see the various stages of the process of ESPTool as it goes through the upload process, however in VSCode-Arduino I don't see this.  How do I set VSCode such that I can see the status and % progress of uploads please?

Many thanks.

JohnboyK, 2023-03-21:
Have you tried in settings of Arduino extension the parameter "Log level"?  You can change this parameter from "info" to "verbose".

maksymovi, 2023-04-22:
I'm experiencing the same issue, enabling verbose log level does show the expected output but its interwoven with a mess of other verbose output, the output OP is requesting is pretty important and preferably should be outputted with standard logging.

Here is the expected output which I pulled from the Arduino IDE

````Sketch uses 259305 bytes (19%) of program storage space. Maximum is 1310720 bytes.
Global variables use 22352 bytes (6%) of dynamic memory, leaving 305328 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/redactedserialport
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:c6:f7:1d:bb:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0004ffff...
Compressed 18960 bytes to 13073...
Writing at 0x00001000... (100 %)
Wrote 18960 bytes (13073 compressed) at 0x00001000 in 0.5 seconds (effective 326.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 367.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 585.1 kbit/s)...
Hash of data verified.
Compressed 259664 bytes to 144099...
Writing at 0x00010000... (11 %)
Writing at 0x0001d72e... (22 %)
Writing at 0x00024dfa... (33 %)
Writing at 0x0002a17b... (44 %)
Writing at 0x0002f4e3... (55 %)
Writing at 0x0003573d... (66 %)
Writing at 0x0003fd7e... (77 %)
Writing at 0x00045968... (88 %)
Writing at 0x0004ae9a... (100 %)
Wrote 259664 bytes (144099 compressed) at 0x00010000 in 2.1 seconds (effective 1008.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
````

Here on the other hand is the output from VSCode when clicking upload.

````
[Starting] Uploading sketch 'project.ino'
Please see the build logs in output path: /Users/maksym/repos/build
Sketch uses 332757 bytes (25%) of program storage space. Maximum is 1310720 bytes.
Global variables use 23244 bytes (7%) of dynamic memory, leaving 304436 bytes for local variables. Maximum is 327680 bytes.
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Cmd+Alt+I"
[Done] Uploading sketch 'project.ino'
```

The big issue with this is when there is an upload error it doesn't tell me useful information, it just says error or failed.

````
[Starting] Uploading sketch 'project.ino'
Please see the build logs in output path: /Users/maksym/repos/build
Sketch uses 344857 bytes (26%) of program storage space. Maximum is 1310720 bytes.
Global variables use 23312 bytes (7%) of dynamic memory, leaving 304368 bytes for local variables. Maximum is 327680 bytes.
Error during Upload: Failed uploading: uploading error: exit status 2
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Cmd+Alt+I"
[Error] Uploading sketch 'project.ino': Exit with code=1
````

From what I understand there is a compile command which properly outputs, but for some reason when running the output command `esptool` the extension isn't dumping the output into the output window or something for the extension. Manually running esptool from cli does show the wanted output without any higher levels of logging and seems to be dumped into standard output. 

Here is my arduino.json, mostly default generated by the extension.

````
{
    "configuration": "JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none",
    "board": "esp32:esp32:esp32",
    "programmer": "esptool",
    "sketch": "project.ino",
    "output": "../build",
    "port": "/dev/redactedserialport"
}
````

Sadly haven't been able to find anything online solving this issue. [Here is one more person experiencing this issue with no solution though.](https://www.reddit.com/r/esp32/comments/11xg3vv/view_esptoolpy_output_during_upload_vscode/), no clue if this is an issue experienced only by esptool or other unloaders.

Steps to replicate (probably):

1. Install the esp32 board library by adding `https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json`.
2. Install the esp32 board package from the board manager.
3. Connect an esp32 board to usb and hopefully have serial pop up correctly.
4. Click upload button in vscode in a sketch.

maksymovi, 2023-04-25:
Anyways this has annoyed me enough that it got me digging within the code. Apparently this isn't a bug, its deliberate design.

It seems not showing output at all for "info level logging" is Microsoft's idea of "info level logging." [In non verbose logging the code apparently runs all commands in verbose mode and then just ignore it all](https://github.com/microsoft/vscode-arduino/blob/9544a5feb30a62b7d762306668716606cccf7c7d/src/arduino/arduino.ts#L811) unless it [starts with "Sketch uses " or "Global variables use ".](https://github.com/microsoft/vscode-arduino/blob/9544a5feb30a62b7d762306668716606cccf7c7d/src/arduino/arduino.ts#L536). To be fair to MS, some of the stuff esptool outputs should probably be dumped to standard error which would show up in the output log, but still.

 I get that logs can be overwhelming sometimes (as is the case in the verbose) but it would be kinda neat to see the upload process, especially to be able to find out at what point during the build/upload process something went wrong.

I really really really like this extension but this is pretty close to a dealbreaker, I wasted hours over a solid week because I had no idea that the random errors I was getting were because the upload port changes sometimes because the upload log/error didn't tell me anything.

laurensV, 2023-09-06:
would also love to have a solution to this problem, at least be able to see upload progress

dbcook, 2023-10-28:
Agree, this is a big problem and deficiency in comparison to ArduinoIDE2.  The device flashing output from esptool needs special handling so that it not suppressed except by deliberate request or config.  As an embedded developer for the last 40 years or so, I can say with high confidence that this info is *always* wanted.

A related bug adding to the annoyance is that the extension sometimes (I'm on Mac) fumbles the serial connection and tries to re-open it when it's already got it open (verified this condition with lsof), leading to yet more unexplained error 2 fails.

luigibrancati, 2023-12-12:
I would also like this

oleg-shilo, 2023-12-19:
I second (third, fourth,....) it.

In this case, the information we are asking to be displayed is not really "log content" but "progress info". There is a huge difference. 

Visual Studio user experience has captured it pretty well. It is a separate visual indication channel for the progress of the solution building and log info at runtime. But VSCode seems still struggling with it a little. 

Building (compiling/uploading) progress feedback is vital. Particularly if there is an uploading problem as it can only be detected after the upload timeout (more than a minute). But progress updates are almost an immediate indication that uploading is going according to the plan.

Please consider implementing it the same way as other IDEs do it for other frameworks.

dalklein, 2024-01-30:
I would also like to see the upload progress in the output window.   I would like to use VSCode for arduino projects, but this is confusing enough to stick with the clunkier (and no Codeium/Copilot coding help) Arduino IDE.   Thanks!