microsoft/vscode-arduino #1624: Built-in Examples missing when installed through marketplace with default settings?

krusion, 2023-03-16:
Hello! 
Wanted to try Arduino with VSCode. Fresh install of windows 11, fresh install of vscode and installed Arduino through the marketplace. I don't have Arduino IDE installed and I didn't install a separate instance of Arduino-CLI. I wanted to use the built in one. 

What I'm missing so far is that I can't seem to find the Built-in Examples. The main folder is there, but it's empty. When I added Arduino Uno as my board and installed the required files, I got the "Examples for Arduino Uno" folder with the 4 folders inside it, yet the "Built-in Examples" is still missing. 
I trying switching to pre-release and even tried installing version 0.5 instead of the current version (v0.6) but still no luck. 

Also a weird question because I saw this on a video, when I open a .ino file the icon shown is for a c++ folder instead of arduino "infinity" logo. Is that normal?

EClemMarq, 2023-05-10:
I see a similar issue on Ubuntu 22.04  vscode v1.78.1, arduino extension v0.6

rnusser, 2023-08-28:
The built in examples are not included with the `arduino-cli`. They can be downloaded from https://github.com/arduino/arduino-examples

However, what I would like to know is, where can we place these examples such that they show up in the "Built-in Examples" menu on the left?

MasterDevX, 2023-10-13:
> However, what I would like to know is, where can we place these examples such that they show up in the "Built-in Examples" menu on the left?

I've been messing around reading the project's code and found out you have to put the `examples` folder inside your `arduino-cli` directory. For example, when using bundled `arduino-cli` on Linux, installing the examples is as simple as doing:
```
git clone https://github.com/arduino/arduino-examples.git
cp -r arduino-examples/examples ~/.vscode/extensions/vsciot-vscode.vscode-arduino-*-linux-x64/assets/platform/linux-x64/arduino-cli/
```
These will show up in the VS Code after:
[image]

blfuentes, 2023-10-26:
> https://github.com/arduino/

this seems to work, thanks!

gpeden, 2024-03-29:
This worked! thanks for the secret handshake!