microsoft/vscode-arduino #1630: Error: unable to locate `compile_commands.json`

avsteele, 2023-03-23:
## Description

If you set the `output` path to something inside the project folder I get errors when trying to verify the sketch.

The error looks like
```
[Starting] Verifying sketch 'testing_error.ino'
Please see the build logs in output path: c:\Users\avste\source\Arduino\testing_error\.out
Error during build: unable to read contents of the source item: open C:\Users\avste\source\Arduino\testing_error\.out\compile_commands.json: The system cannot find the file specified.
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'testing_error.ino': Exit with code=1
```

- The first time you verify, or if you delete the existing output folder, it does not give this error. 
- Occasionally (maybe 1/5 times) it will build without this error
- If the output folder is where outside the project folder (e.g. `c:\temp\`) you will not get this error (ever as far as I've seen)
- After you get the error the `compile_command.json` file will be empty except for a `[]`

## Steps to replicate 
Make a new project. Board= `Uno`, programmer = `AVR ISP`

sketch file
```cpp
#include <Arduino.h>
void setup(){}
void loop(){}
```

`arduino.json`
```json
{
    "board": "arduino:avr:uno",
    "programmer": "avrisp",
    "sketch": "testing_error.ino",
    "output":".out"
}
```

in `arduino.json` set the output folder to something relative (.e.g `.out`). Compile once (will complete ok.  If you try it again, you will get the error (~80% of the time). If you get the output folder somewhere else you do not get the error.

GregTerrell, 2023-04-13:
Also seeing this. If the "empty" compile_commands.json file is deleted the build succeeds. 

I have relied on the .build folder (located as a sibling to  the .vscode folder in the project) to speed builds. If you are working with a large number of sketch folders it is difficult to have unique locations for the build output **outside** of the project folder. Possibly allowing a ${} variable inside the arduino.json output parameter would be an alternative.

mikalhart, 2023-04-13:
I'm seeing this as well, and concur with @GregTerrell that deleting the (almost) empty compile_command.json file makes the build succeed (but then you get a full rebuild and you've lost all the advantage of having "output" in arduino.json).

FWIW, my "**output**" folder is currently "**./.output**".

37electrons, 2023-04-15:
Found this discussion since was seeing exactly the same issue. My setting in 'arduino.json' was  **"output": ".//.build"**

After it was changed to  **"output": "./build"**  everything started working every single time.

Not sure what the root cause of the problem was though. Perhaps the directory name starting with a dot.  I am on Win10, fwiw.

GregTerrell, 2023-04-16:
Concur with @37electrons. The behavior seems tied to the . prefix on the build directory **if it is in the project directory**. If the  .build is in the parent directory, seems to work ok.

I have been using .build to follow the .vscode convention and simplify my gitignore rules.

avsteele, 2023-05-08:
The error persists if the output directory in a subfolder, even if it does not start with a dot, **if** the name of the folder **contains** the word `out` (at least).

These all result in the error

```json
"output": "out"
"output": "output"
"output": "outzz"
"output": "zzout"
"output": ".output"
"output": ".//.output"
"output": ".build"
"output": ".//.build"
```

These do not
```
"output": "build"
"output": "./build"
```