microsoft/vscode-arduino #1635: Serial monitor fails to connect and disconnect the port

TennisRunner, 2023-03-29:
After the first time serial monitor starts monitoring. It fails to reconnect ever again.

Expected behavior:
Click Arduino: Upload.
Serial monitor disconnects.
Arduino build uploads.
Serial monitor reconnects.

Actual behavior:
Click Arduino: Upload.
Serial monitor claims to disconnect
Arduino build uploads.
Baud rate reverts back to 115200 from 9600
Serial monitor says it failed to open the port
Clicking "Start monitor" again results in the same failed to open message.

If click Arduino: Upload again, receive message:
`avrdude: ser_open(): can't open device "\\.\COM7": Access is denied.`

Clicking close serial monitor button does not disconnect the port.

Issue persists until vs code is restarted.

This is the log in the serial monitor:

```
---- Opened the serial port COM7 ----
this is my arduino println
---- Closed the serial port COM7 ----
---- Closed the serial port COM7 ----
---- Opened the serial port COM7 ----
**** Failed to open the serial port COM7 ****
```

BlackholeBun, 2023-04-09:
I have the same issue. Have to close and reopen code every time I upload.

gcampbell-msft, 2023-04-10:
@BlackholeBun  @TennisRunner Hi! Thanks for reporting the issue.

I was wondering if you could do some investigation for me to try and diagnose the issue. Have you noticed this issue when you use the Serial Monitor independently? 

Additionally, @TennisRunner, in regards to the baud rate changing, are you reselecting the baud rate when you try and upload? Every time you upload you should be prompted with a choice for baud rate, if you don't select this a default will be chosen. 

I look forward to getting these answers so that I can investigate and hopefully resolve the issue!

closed 2023-05-17 by github-actions[bot]