microsoft/vscode-arduino #1637: Failed to open the serial port /dev/ttyACM1

EGalahad, 2023-04-15:
<details><summary>Board</summary>
<p>
Arduino UNO R3
</p>
</details> 

<details><summary>OS</summary>
<p>
Arch Linux
kernel: 6.2.8-arch1-1
</p>
</details> 

<details><summary>Sketch</summary>
<p>

```c++
void setup()
{
  Serial.begin(9600);
  Serial.println("Hello World!");
  // pinMode(ledPin, OUTPUT);
}

void loop() {
  Serial.println("Hello From loop()!");
  delay(1000);
}
```

</p>
</details> 

<details><summary>Error</summary>
<p>
In serial port tab, 

```
**** Failed to open the serial port /dev/ttyACM1 ****
```

</p>
</details> 
<details><summary>arduino.log</summary>
<p>

```json
{"notification":"Serial Monitor API was not retrieved. You may not have the most recent version of the Serial Monitor extension installed.","showUser":true,"telemetry":true,"message":"UndefinedSerialMonitorApi","errorLine":"src/serialmonitor/serialMonitor.js:167:69)","level":"error","timestamp":"2023-04-15T01:28:13.063Z"}
```
</p>
</details> 

<details><summary>Additional Info</summary>
<p>
1. Arduino IDE v2 can use serial port.
2. Arduino Extension use bulit-in arduino-cli.
</p>
</details>

gcampbell-msft, 2023-05-22:
@T0mHutter To clarify, when I mentioned v0.8.0, I was referring to the vscode-serial-monitor extension, could you confirm that you see that version for that extension?

T0mHutter, 2023-05-22:
I did mean the serial port extension

gcampbell-msft, 2023-05-22:
What version of `ms-vscode.vscode-serial-monitor` do you see in the extensions marketplace in VSCode? I currently see `v0.9.0` for the release version and `v0.9.230516001` for pre-release.

T0mHutter, 2023-05-22:
I See 0.7.0 but let me check again if I have Wi-Fi

T0mHutter, 2023-05-23:
it seems as if the version "changes" when reloading to get everything to work fully so it shows v.0.9.0 as a pre release then when i reload it changes to 0.7.0 again and the other way around

gcampbell-msft, 2023-05-23:
@T0mHutter Great! I'm glad you were able to solve your issue. Closing.

closed 2023-05-23 by gcampbell-msft