microsoft/vscode-arduino #1638: Switched to arduino-cli, no more warnings

coffeemachine-ftq, 2023-04-16:
Since I switched to arduino-cli, I no longer get warnings for my code. It looks like we need a way to pass to following arguments to arduino-cli
```
--warnings all
```
See [arduino-cli compile documentation](https://arduino.github.io/arduino-cli/0.32/commands/arduino-cli_compile/)

Since compiler warnings are an essential developer tool, they should be enabled by default. When I use arduino-cli from the command line with `--warnings all` I get the output I want.

I tried other alternatives using `buildPreferences` in `arduino.json`, none of them was successful:

- `["compiler.warning_level", "all"]` doesn't produce any warning (seems that was the way to go with the IDE).
- `["build.extra_flags", "-Wall"]` doesn't produce any warning and causes the verification to fail with errors in the arduino core code.
- `["compiler.cpp.extra_flags", "-Wall"]` doesn't produce any warning.
- `["warnings", "all"]` doesn't produce any warning.

jmailloux, 2023-04-24:
I see this issue as well.  I have this in arduino.json:
    "buildPreferences": [
        ["build.extra_flags", "-Wall -Wextra"]
      ]

I found that actually -Wall -Wextra do get added in the compiler command.  However, they don't do anything because -w (which inhibits all warnings) is inserted earlier on in the compiler invocation.
So as coffeemachine-ftq says, looks like we need a way to pass --warnings all to arduino-cli.

jmailloux, 2023-04-24:
This is the only thing keeping me from using arduino-cli in VSCode.

rajiv-obhcare, 2023-06-27:
Is there a known workaround for this?

TheAlgorythm, 2023-11-22:
@rajiv-obh i have created a workaround which works with the current megaavr core and presumably most others.

```console
sed -i 's|compiler.warning_flags=-w|compiler.warning_flags=-Wall -Wextra -pedantic|g' /root/.arduino15/packages/arduino/hardware/**/**/platform.txt
sed -i 's|=-w |=|g' /root/.arduino15/packages/arduino/hardware/**/**/platform.txt
```

The warning level can be adjusted by changing `-Wall -Wextra -pedantic` in the first command to the supposed one like `-W`.