microsoft/vscode-arduino #1646: CLI Upload Fails with unknown --build-path flag

WhoShotCupidd, 2023-05-20:
Ran on the prerelease and current release of extension.

error: unknown flag: --build-path
Usage:
  arduino-cli upload [flags]

Examples:
  /Users/christiaan/.vscode/extensions/vsciot-vscode.vscode-arduino-0.6.230420001-darwin-arm64/assets/platform/darwin-arm64/arduino-cli/arduino-cli.app upload /home/user/Arduino/MySketch

Flags:
      --board-options strings        List of board options separated by commas. Or can be used multiple times for multiple options.
      --discovery-timeout duration   Max time to wait for port discovery, e.g.: 30s, 1m (default 1s)
  -b, --fqbn string                  Fully Qualified Board Name, e.g.: arduino:avr:uno
  -h, --help                         help for upload
      --input-dir string             Directory containing binaries to upload.
  -i, --input-file string            Binary file to upload.
  -p, --port string                  Upload port address, e.g.: COM3 or /dev/ttyACM2
  -m, --profile string               Sketch profile to use
  -P, --programmer string            Programmer to use, e.g: atmel_ice
  -l, --protocol string              Upload port protocol, e.g: serial
  -t, --verify                       Verify uploaded binary after the upload.

Global Flags:
      --additional-urls strings   Comma-separated list of additional URLs for the Boards Manager.
      --config-file string        The custom config file (if not specified the default will be used).
      --format string             The output format for the logs, can be: text, json, jsonmini, yaml (default "text")
      --log-file string           Path to the file where logs will be written.
      --log-format string         The output format for the logs, can be: text, json
      --log-level string          Messages with this level and above will be logged. Valid levels are: trace, debug, info, warn, error, fatal, panic
      --no-color                  Disable colored output.
  -v, --verbose                   Print the logs on the standard output.

unknown flag: --build-path
[Warning] Failed to generate IntelliSense configuration.
[Error] Uploading using Arduino CLI sketch 'main/main.ino': Exit with code=1

emabrey, 2023-08-02:
This is being caused here:

 https://github.com/microsoft/vscode-arduino/blob/9544a5feb30a62b7d762306668716606cccf7c7d/src/arduino/arduino.ts#L718
 
 I get similar behavior if I specify a `buildPreferences` key in `arduino.json` (see https://github.com/emabrey/protogen/commit/c0074520dc5d0fd90288ad3bbddefa33110873bf).
 
 This bug seems to be related to passing invalid arguments to the `arduino-cli` tool. The `arduino-cli upload` command doesn't support an option by the name of `--build-path` (though it does support an option called `--input-dir` to specify the directory containing the uploadable binary files).

philipmcgaw-cpi, 2023-10-21:
@emabrey have you got a work around, or come up with a way of getting this to work? is it only an issue on Mac, or is it the same issue on Windows and Linux?

rrainey, 2024-01-20:
@philipmcgaw-cpi it's an issue on Windows 11, too.

Joshuabaker2, 2024-01-28:
There's definitely an issue with the code, but I was able to work around it by using the command `Arduino: Upload` as opposed to `Arduino CLI: Upload`.