microsoft/vscode-arduino #1651: sketch path in arduino.json not case sensitive

kaffepanna, 2023-06-12:
On a case sensitive file system you cannot have any capital letter in your sketches filename.

For example creating a sketch named `Example.ino` with `.vscode/arduino.json`

````json
{
    "sketch": "Example.ino",
    "configuration": "mode=off",
    "board": "arduino:megaavr:nona4809",
    "output": "build"
}
````

will result in:

```
[Starting] Analyzing sketch 'Example.ino'
Please see the build logs in output path: /home/patrik/sources/example/build
Error opening sketch: main file missing from sketch: /home/patrik/sources/example/example.ino
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch 'Example.ino': Exit with code=1
```

changing the file to all lower case such as `example.ino` will work regardless of the case in `arduino.json`.

Tested with v0.6.0 and v0.6.230420001 on linux

kaffepanna, 2023-06-12:
After doing some digging found this in Arduino CLI

[arduino sketch.go](https://github.com/arduino/arduino-cli/blob/master/arduino/sketch/sketch.go#L51)

```
// New creates an Sketch instance by reading all the files composing a sketch and grouping them
// by file type.
func New(path *paths.Path) (*Sketch, error) {
	if path == nil {
		return nil, fmt.Errorf(tr("sketch path is not valid"))
	}

	path = path.Canonical()
	if exist, err := path.ExistCheck(); err != nil {
		return nil, fmt.Errorf("%s: %s", tr("sketch path is not valid"), err)
	} else if !exist {
		return nil, fmt.Errorf("%s: %s", tr("no such file or directory"), path)
	}
```

`path.Canonical()` is described as such in [arduino go-path-helper](https://github.com/arduino/go-paths-helper/)

>  // Canonical return a "canonical" Path for the given filename.
    // The meaning of "canonical" is OS-dependent but the goal of this method
    // is to always return the same path for a given file (factoring out all the
    // possibile ambiguities including, for example, relative paths traversal,
    // symlinks, drive volume letter case, etc).

kaffepanna, 2023-06-12:
Soo after some more digging it seems like the "main" sketch file needs to have the same name as the parent folder (with casing and all). But then the question becomes. why the need to specify a sketch file in visual studio?

Sorry for my ignorance but this is day two of me tring to figure out the arduino.. build system?

MartinEmrich, 2023-07-14:
I have the same isssue. @kaffepanna have you found a solution?

I would love to see that either the convention "<dirname>/<dirname>.ino must be the main file" is documented, or the "sketch" parameter respected.