microsoft/vscode-arduino #1653: Solution for including librararies, or setting Arduino CLI config options

wilg, 2023-06-16:
This extension has almost everything in place, but it seems like there is still no way to set any of the [Arduino CLI config options](https://arduino.github.io/arduino-cli/0.20/configuration/) which would allow you to set `ARDUINO_DIRECTORIES_USER` which would allow you to pull in libraries from your project folder.

I tried using a `prebuild` command to `export ARDUINO_DIRECTORIES_USER="$(pwd)/arduino"` but unfortunately that seems to have a different environment than the one that ultimately runs the Arduino CLI.

Is there a workaround I'm missing or is a change needed here? 

Seems to be a common problem:

- https://github.com/microsoft/vscode-arduino/issues/1619
- https://github.com/microsoft/vscode-arduino/issues/1614
- https://github.com/microsoft/vscode-arduino/issues/1602
- https://github.com/microsoft/vscode-arduino/issues/1595
- https://github.com/microsoft/vscode-arduino/issues/1341
- https://github.com/microsoft/vscode-arduino/issues/1254

AJ-Work-Smoothie, 2023-10-05:
Has this seriously not been solved yet? I cannot find the answer anywhere

AJ-Work-Smoothie, 2023-10-27:
@armando-fandango @mgiglia92 I was able to correct the problem with these instructions! https://github.com/microsoft/vscode-arduino/issues/1614

mgiglia92, 2023-10-27:
> @armando-fandango @mgiglia92 I was able to correct the problem with these instructions! #1614

I was trying to do everything without the IDE but it seems we still need the Arduino/libraries directory. I just symbolically link my custom libs into here as a solution for now. It's unfortunate that Arduino doesn't have a simplified feature for this. maybe it's in 2.0 but I'm trying to move away from arduino in general to be more "robust" in my software dev.

Thank you though!

Henry-Jones, 2023-11-01:
Technically yes, version 2.0 is able to have a project centric include folder. It can be set in multiple ways (environment variable, config file, command line argument). Also the Arduino executable can report the currently used folder and in which way it was set.

Both features are not used from the VSC plugin, it only considers the standard paths.

waqasahmad531, 2023-12-05:
Answered it already at another place but posting it here in case someone does not go to 1614. 
The solution that worked for me was to open the the C:\Users\<username>\AppData\Local\Arduino15\arduino-cli.yaml file and update the user directory to the installation path of arduino which in my case was: C:\Program Files (x86)\Arduino

vmajor, 2023-12-19:
Please fix this...I lost hours figuring out why VSC cannot find my libraries. That warning about moving away from Arduino IDE really forces this issue, and it is ...not good that the problem with paths is both old, and still unresolved.

paaspaas00, 2024-01-10:
This issue is really a blocker for me, could you prioritize it? Please

Maljanen, 2024-02-21:
To be honest it is bit unnecessarily confusing when Prerequisites lists legacy Arduino IDE or Arduino CLI (with one coming bundled with extension), when having both and using bundled CLI still means one seems to for example still need to manually figure out what SPI.h version they should be using for their hardware, then manually add paths to it, despite there being option to select what board they are using.

Also libraries that are installed with ArduinoIDE library manager do not seem to be found, but also ones installed in vscode with extension's library manager are not found. All resulting in include error, unless one apparently points manually to all of them in c_cpp_properties.json.

This might be small thing when someone has routine, but starting, it is annoying day of searching for posts online, with nearly all of them mentioning "SPI.h is standard library/dependand on board, so it all happens automatically", when in fact it does not seem to, or them pointing to "yeah libraries are in c:\program files (x86)\arduino\..." when they actually do not seem to be.
Installed ArduinoIDE about 2 months ago, and absolutely nothing got installed to that path.

Have went and looked that I have use ArduinoCLI as true in user and project settings, and arduino.path and arduino.commandPath empty/unset.

Everything else looks pretty much perfect this far, but having constant #include error is bit annoying and worrying.
I am still quite convinced it is something really small in settings or something I am missing that results in this, but also thinking should I actually be defining that by hand, but at same time wondering what ultimately is point in board selection being included if I should then still be doing things related to it manually.

Also if we have library manager, why libraries installed with it wont be found, even if "Include Library" button is pressed, does that button even do anything, since I am not seeing any changes to anything, what is it even supposed to do?

I am guessing that (hopefully) 15minutes after commenting this I will realize or find something that just fixes everything and makes this comment look foolishly stupid, but oh well guess I gotta comment this to find out if that will happen.