microsoft/vscode-arduino #1657: Cannot open any serial port even serial monitor

S-i-l-v-e-t, 2023-07-02:
Serial Monitor API was not retrieved. You may not have the most recent version of the Serial Monitor extension installed.
Reinstalled newest version and not solved.
too many configurations, i give up

C0dex73, 2023-07-03:
Same issue, i have the 0.10.0 - 2023-06-07 installed

S-i-l-v-e-t, 2023-07-06:
> What version of the Serial Monitor extension do you have installed? Thanks!

i have the 0.7.0. and my Visual Code said its the newest version :(

BloodyRain2k, 2023-08-29:
> What version of the Serial Monitor extension do you have installed? Thanks!

Tried Pre-Release `v0.7.221230001` because `v0.7.0` that came with the Arduino Extension didn't work either.

S-i-l-v-e-t, 2023-09-03:
thank for your constructive advice. maybe i'll get the extension back someday and try it my self.
to be honest, i even sell my Arduino uno and turn my whole project to STM32 with a raspyberryPi as the PC

closed 2023-09-03 by S-i-l-v-e-t