microsoft/vscode-arduino #1664: Output path is not specified. Unable to reuse previously compiled files. Build will be slower.

bvking, 2023-07-11:
Hello.
I would like to execute Arduino Ide directly from Visual Code but I have this message error.
" Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
Memory Usage on Teensy 4.1: ".
It 's not due to teensy 4.1 because I can excuse my program from Arduino Ide to control the teensy 4.1.
Many thanks fr trying to help ;)

jakeread, 2023-07-20:
this is not an error, it is simply a note that if you include in ".vscode/arduino.json" an output directory, a-la:

```json
{
    "configuration": "flash=2097152_0,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Disabled,dbgport=Disabled,dbglvl=None,usbstack=picosdk,ipbtstack=ipv4only,uploadmethod=default",
    "board": "rp2040:rp2040:seeed_xiao_rp2040",
    "sketch": "maxl-stepper.ino",
    "port": "COM23",
    "output": "./build" 
}
```

then arduino-vscode will be able to use that directory to stash data from previous compiles, to speed it up

Xpiravit, 2023-08-03:
Is there a way to have a default output directory so that I don't have to add it to all json files?
Where can I edit the default form of my arduino.json?

BlindSanji, 2023-10-05:
I also want to know if it's possible to set a default for the output.

shtirlic, 2024-04-05:
+1 to make default