microsoft/vscode-arduino #1665: Main file missing from sketch

Lisoveliy, 2023-07-12:
Hi. I think, it's a bug.  After initializing arduino "project" in arduino.json i have that data. 
[image]
File main.ino exists
[image]
But after try to compile i have that error.
[image]
Entry point set in arduino.json as "main.ino" but extension also trying to find sketch named like folder. Please fix that, or maybe i mistaked somewhere?

GimpMaster, 2023-07-15:
I'm getting the same issue as well. Any suggestions?

GimpMaster, 2023-07-15:
It seems like if you rename your sketch to match the folder name it works, Seems like a bug.

uztbt, 2023-07-22:
Is there any configuration parameter to specify a custom main file name?

Lisoveliy, 2023-07-22:
I think, it is. It named "sketch" but it doesn't working.

jdsalmonson, 2023-09-15:
Having this issue as well.  GimpMaster's solution works for me, but have we determined if this is a bug?

spinpwr, 2023-10-09:
Using the same name for the folder (i.e. "project") and the main file is the same what the Arduino IDE does, right?
That could be a reason, but still we have a config for the main "sketch" which doesn't work.. so it's a bug for me

OriginalCREart, 2023-10-16:
> Using the same name for the folder (i.e. "project") and the main file is the same what the Arduino IDE does, right? That could be a reason, but still we have a config for the main "sketch" which doesn't work.. so it's a bug for me

It turns out that the main sketch **MUST** be the same name of the container folder, I was having a file called **WifiManager.ino** inside a folder called **WifiManager** [**WifiManager\WifiManager.ino**], I renamed the **WifiManager.ino** to **Ultrasonic.ino**, and the verification/compilation/upload was failing, after reading @spinpwr comment, I renamed the folder to be as the same as the new sketch name **Ultrasonic.ino** but without the .ino extension and I ended up having [**Ultrasonic\Ultrasonic.ino**] and everything worked just fine and I was able to verify/compile/upload and rebuild IntelliSense configuration [_shortcut -> "Ctrl+Alt+I"_].

Thanks

HeatPhoenix, 2023-10-16:
Agreed, the .json file should determine the main file. Not the folder name.

manaspace, 2023-10-26:
Do not use Arduino Cli. Use Arduino Path and command targeting to your arduino ide folder, solves the error.

alelavoie, 2024-01-05:
Looks like a bug indeed. To fix the issue, I simply created an empty `.ino` file with the same name as the project directory in the root of the project: `My Project/My Project.ino`

QuentinHsu, 2024-02-20:
> Agreed, the .json file should determine the main file. Not the folder name.

I agree too! Otherwise, what's the point of the `sketch` field in the generated `.vscode/arduino.json`?!

msingh0101, 2024-03-09:
so is this gonna get marked as a bug or not? 
Also, why cant the extension detect the working file rather than inferring it or using arduino.json (when fixed)?

vadixidav, 2024-03-19:
Still an issue.

Facul-101, 2024-07-17:
When will it be fixed?