microsoft/vscode-arduino #1669: How do I add custom *.h library Include Paths?

jleaders, 2023-07-23:
It seems like this extension follows in the path of the Arduino IDE, which doesn't have an industry standard C++ build system.
Therefore a simple thing like setting up an include path is not possible, so larger projects become infeasible.

jleaders, 2023-07-23:
I created a GUI tool to manage custom include paths, it even parses out the include paths from CMakeLists.txt to some level, but I would need some help open-sourcing it. It works by creating symlinks to your active *.ino file from include paths that are stored in a new YAML file call `.arduinovscde` 

Here's what the GUI looks like:

[image]

jleaders, 2023-07-30:
Until we get a tool like that open sourced, let's update the Readme.md with instructions on how to add libraries to the VS Code extension, either as an Arduino library, or a classic C++ lib with include path of .h files

2bam, 2024-03-29:
Seems like this setting (in the screenshot) doesn't even work, which is misleading.
Not sure this setting is Arduino-extension related, but could be probably read from the vscode api.

[image]