microsoft/vscode-arduino #1677: Chinese Language Support Incorrect

Tianming-Wu, 2023-08-29:
The "output" terminal writes some of the things in an incorrect codec. As below.
[image]

Tianming-Wu, 2023-08-29:
I think it used gbk and VSCode used UTF-8

yffbit, 2023-12-15:
use config file and set locale to en, restart vs code
```
%userprofile%\.vscode\extensions\vsciot-vscode.vscode-arduino-0.6.0-win32-x64\assets\platform\win32-x64\arduino-cli\arduino-cli.exe config init -v
%userprofile%\.vscode\extensions\vsciot-vscode.vscode-arduino-0.6.0-win32-x64\assets\platform\win32-x64\arduino-cli\arduino-cli.exe config set locale en -v
```