microsoft/vscode-arduino #1694: [Error] Verifying sketch {"code":null}

themaninthesuitcase, 2023-11-27:
I am getting the following output when trying to build Arduino projects on my Mac.

I have used the blink sample code to exclude anything in my code.

```
[Starting] Verifying sketch 'Blink.ino'
Please see the build logs in output path: /Users/chris/Library/Mobile Documents/com~apple~CloudDocs/ArduinoOutput/Blink
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'Blink.ino': {"code":null}
```

Checking the quoted output path shows it failed to output anything.

I also tested without "output" set incase is there was an issue with the path into my iCloud Drive.  It made no difference.

The exact same code compiles when using the Arduino IDE and when using VS Code Arduino on Windows.  

Versions: 
VS Code 1.84.2
vscode-arduino 0.6.0
Arduino IDE 2.2.1
macOS 14.1.1 (M1)
Windows 10 22H2

Targeting Teensy 4.1, but swapping to a Nano also fails.

bringert, 2023-12-05:
I'm seeing the same symptoms. 

Boards:
Arduino Nano 33 BLE
Arduino Nano 33 IoT

vscode-arduino: v0.6.0
VS Code: 1.84.2 (Universal)
MacOS: 14.1.2 (23B92)

pauls-z, 2023-12-05:
This issue is already registered and can use https://github.com/microsoft/vscode-arduino/issues/1681#issuecomment-1747890450 as a workaround for now.