microsoft/vscode-arduino #1706: Can't upload sketches to Arduino R4 Wifi

imfatant, 2024-01-28:
Type: <b>Bug</b>

Using VS Code in Windows 10/11 you simply can't upload sketches to  the Arduino R4 Wifi. R4 Minima is fine though. Typical output:

[Starting] Uploading sketch 'GameOfLife.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
Sketch uses 54640 bytes (20%) of program storage space. Maximum is 262144 bytes.
Global variables use 7408 bytes (22%) of dynamic memory, leaving 25360 bytes for local variables. Maximum is 32768 bytes.
No device found on COM7
Error during Upload: Failed uploading: uploading error: exit status 1
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
[Error] Uploading sketch 'GameOfLife.ino': Exit with code=1

Extension version: 0.6.0
VS Code version: Code 1.85.2 (8b3775030ed1a69b13e4f4c628c612102e30a681, 2024-01-18T06:40:10.514Z)
OS version: Windows_NT x64 10.0.22621
Modes:

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz (16 x 3600)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: enabled_on<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>video_decode: enabled<br>video_encode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled<br>webgpu: enabled|
|Load (avg)|undefined|
|Memory (System)|31.91GB (19.52GB free)|
|Process Argv|--crash-reporter-id 5cd84fa0-7140-4f06-b37c-544d8b014ffd|
|Screen Reader|no|
|VM|0%|
</details>

m4x74, 2024-06-15:
I have the same problem. This was also reported in [this thread of the Arduino Forum](https://forum.arduino.cc/t/vs-code-wont-upload-to-arduino-uno-r4-wifi/1216533)

Joaquin-Penguin, 2024-08-09:
Bump! If you fix this I will buy you a beer