microsoft/vscode-arduino #1722: Programmer setting is ignored, causing upload to fail

kaysond, 2024-03-11:
I have a USBASP programmer, and my `arduino.json` has `"programmer": "usbasp"`. I have no choice but to set a COM port, or the extension won't allow me to upload In this case I used `"port": "port"` as suggested in https://github.com/microsoft/vscode-arduino/issues/630#issuecomment-641699390. The `programmer` setting then seems to be ignored, as `avrdude` tries to upload via the com port even though I've selected my programmer.

```
"C:\Users\aram\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\aram\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -V -patmega328p -carduino "-Pport" -b57600 -D "-Uflash:w:y:\git\hawgtrophy\build/hawgtrophy.ino.hex:i"
```

kaysond, 2024-03-11:
This looks like a pretty easy fix, and I'd be happy to submit the PR if you'll merge it.

Update [src/arduinio/programmer.ts](https://github.com/microsoft/vscode-arduino/blob/main/src/arduino/programmer.ts) to parse `*.communication` from `programmers.txt`.

Update 
https://github.com/microsoft/vscode-arduino/blob/4ecf9e431708f44be238f4a3974109dab6f7b2ba/src/arduino/arduino.ts#L629-L644

and

https://github.com/microsoft/vscode-arduino/blob/4ecf9e431708f44be238f4a3974109dab6f7b2ba/src/arduino/arduino.ts#L651-L662

to only call `selectSerial` and push the argument if `*.communication === "serial`

itodirel, 2024-03-25:
Hi, we see this as a feature request. At this time we don't plan on supporting this scenario.

kaysond, 2024-03-25:
> At this time we don't plan on supporting this scenario.

You don't plan on supporting USB programmers?

itodirel, 2024-03-25:
We don't plan on supporting the scenario you outlined. If you connect an Arduino via USB and use the built-in bootloader, that's the only scenario we support and that should work. If you have issue with that scenario, let us know.

kaysond, 2024-03-25:
> We don't plan on supporting the scenario you outlined. If you connect an Arduino via USB and use the built-in bootloader, that's the only scenario we support and that should work. If you have issues with that scenario, let us know.

That's disappointing. Why such limited support? There seems to be a lot of interest from the community to improve the extension, but not so much from Microsoft to accept the help...

CodeAsm, 2024-04-20:
Please support this scenario. the core of teh arduino is being able to programm arduino boards. If the bootloader is missing, using a usbasp, same for debugging

buiapp, 2024-07-08:
Please support this. Because of this missing feature my team can't completely ditch the Arduino IDE.