microsoft/vscode-arduino #1751: Does not work on macOS Sonoma (M1 Mac. Mini)

JohnHind, 2024-06-26:
I have spent ages trying to get this working on my Mac, although I have it working well on my Windows PC.

Today, just to be sure I was not doing anything stupid, I completely nuked my Mac and reloaded the OS, the Arduino IDE, VS Code and the Arduino extension. It prompted me to use the bundled Arduino CLI and I gave it permission. I created and opened an empty new folder on the desktop and executed 'Arduino: Initialize' and specified the same name as the folder for the '.ino' file. I selected a board at random and clicked the 'verify' button with the following result:
```
[Starting] Verifying sketch 'test.ino'
Please see the build logs in output path: /Users/johnhind/Desktop/test/build
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'test.ino': {"code":null}
```
The only thing in the specified 'test' directory is the 'test.ino' sketch - no 'build' directory, no logs.

When I restart VS Code, I get the following 'helpful' message:
```
[Starting] Update package index files...
dummy
[Error] Exit with code=null

[Starting] Analyzing sketch 'test.ino'
Please see the build logs in output path: /Users/johnhind/Desktop/test/build
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch 'test.ino': {"code":null}
```
Apart from not appreciating being addressed as 'dummy', this seems to indicate that none of the CLI calls are working.

Has anyone got this working on an Apple Silicon Mac? If so, what might I be doing wrong?

JohnHind, 2024-06-27:
Further to this, I discovered that after the re-install above, the Arduino IDE did not work either: verify gave errors about unsupported CPU. I had to manually install rosetta: 'sudo software update --install-rosetta'. After a reboot, the Arduino IDE compiles fine, but disappointingly, no change in VS Code! Except that a new board URL I entered in Arduino IDE now shows up in Board Manager in VS Code, so they seem to be sharing the same data storage.

lsl233, 2024-07-03:
i have the same issue.