microsoft/vscode-arduino #1757: VS Code Arduino Extension Deprecation

gcampbell-msft, 2024-08-16:
Hi VS Code Arduino users, we make this announcement to inform you that we are deprecating the Arduino extension. We recommend existing customers to use the [Arduino IDE software](https://www.arduino.cc/en/software)

ledgerama, 2024-08-18:
Well that sucks :(

DRSDavidSoft, 2024-08-18:
@gcampbell-msft Please don't do this, even though the official IDE is just [Theia](https://github.com/eclipse-theia/theia) at this point with the [Monaco editor](https://github.com/microsoft/monaco-editor), it sucks for me because I want to use all my cool VS code extensions in there but I can't!

I know @Microsoft might not be interested any more to maintain this, although I wish the Arduino team would pick it up.

In any case, thank you the team at Microsoft. It's been a great journey. I wish you didn't deprecate it. :(

VirensCn, 2024-08-20:
this is really bad news.

CiberNin, 2024-08-20:
#frothing with rage

FrightRisk, 2024-08-20:
Are you out of your minds???? This is insane. Please reverse this and allow other people admin privileges to keep maintaining it.

avsteele, 2024-08-20:
Very surprised to see this. I think if MS provided a rationale it would help people to understand the decision. 

The Arduino IDE is not close to a replacement. The PlatformIO extension for VScode is a good alternative, but the MS extension was the best fit for simpler projects.

AJ-Work-Smoothie, 2024-08-20:
Please don't do this!! The Arduino IDE is NOT a suitable replacement for vsc.

AJ-Smoothie, 2024-08-20:
I agree with @AJ-Work-Smoothie , please don't do this!!

thelastoutpostworkshop, 2024-08-20:
Very sad announcement

escaner, 2024-08-20:
This is really bad news :(

wbadry, 2024-08-20:
I don't know who made the decision. I hope the team will revise it. That extension is so good in many ways beyond the Arduino IDE.

quinn-n, 2024-08-21:
> Very surprised to see this. I think if MS provided a rationale it would help people to understand the decision.
> 
> The Arduino IDE is not close to a replacement. The PlatformIO extension for VScode is a good alternative, but the MS extension was the best fit for simpler projects.

+1 the Arduino IDE is nowhere near close to a replacement for VSCode

mej558, 2024-08-21:
what? this can't be right? Someone please sort this out.  Arduino IDE is a great place to start, but doesn't compare to VSCode to get stuff done.

athm15, 2024-08-21:
not good news since vs code editor is infinitely better

alka79, 2024-08-21:
2.2M downloads and dropped. Bad news for a lot of people :-(
Arduino IDE 2.x is not (yet) a reasonable alternative.

dev-markus, 2024-08-21:
If you're not going to continue dev on it, then at least hand the project over to other developers. It can't be that a functioning extension that really makes sense simply stops.

sjitn, 2024-08-21:
Agreeing with everybody here... PLEASE don't do this. Adiuno IDE is no replacement for VS Code. Pretty pretty please...

FrightRisk, 2024-08-21:
> What are the steps if someone wants to take over the maintenance ?

They would have to allow someone to have maintainer privileges here, or someone very industrious would have to fork it and take over the maintenance. Hopefully, Microsoft will see how bad an idea this is and change their mind.

gitisacatsname, 2024-08-21:
culture changing erverywhere

benreu, 2024-08-21:
What is up, anyway? Espressif drops support for arduino-espressif32 for VS Code, then Microsoft drops support for Arduino in VS Code. The official Arduino software doesn't cut it for heavy developers like myself that spend hours per day compiling, debugging, and changing code for IOT devices. What next, VS Code gets dropped maybe?

chicodaman, 2024-08-21:
Completely agree with the posters above. 

VSCode+Arduino extension is a wonderful middle ground between Arduino IDE 2.X (quick and dirty but super simplistic) and PlatformIO (powerful but plenty of gotcha's -- things sometimes go wrong and can be very difficult to diagnose, e.g. library issues).

Please someone, just keep the Arduino extension functional for those of us who like the middle-ground sweet spot!

KN1HWK, 2024-08-22:
No! The arduino plug in for VSCode had a few bugs that were easy enough to work around but it's still a WONDERFUL tool for Arduino development. The arduino IDE doesn't integrate with Git repos and there are tons of other super useful plugins for VSC that make Arduino dev so pleasant in VSCode. Please don't do this!

alka79, 2024-08-22:
you may have noticed that someone started a community maintained extension
https://marketplace.visualstudio.com/items?itemName=vscode-arduino.vscode-arduino-community

michaeljon, 2024-08-22:
It's the entire VSCode ecosystem that makes the Arduino plug-in far more powerful than the dedicated IDE. Git, formatters, HTML servers, markdown, hell just being able to actually maintain some file structure is hugely better.

chicodaman, 2024-08-22:
> you may have noticed that someone started a community maintained extension
> https://marketplace.visualstudio.com/items?itemName=vscode-arduino.vscode-arduino-community

I see that it hasn't been updated since 3-23. Hopefully now that the Microsoft extension is deprecated there will be renewed activity on that project, at least to maintain ongoing functionality.

davidlmorris, 2024-08-23:
This is very sad.  The Arduino IDE just isn't up to the job, and PlatformIO is behind in its support for the latest Esp32 libs.  This was my major reason for using vscode in the first place.  It is a huge backward step.

artificialfintelligence, 2024-08-23:
Great... *Just* after I got into this hobby. 😞

LuWroblewski, 2024-08-23:
nooooooooooooo 😭

rvpeterson, 2024-08-24:
This is a bit of a nightmare. The native Arduino IDE doesn't support subfolders. Hopefully someone from the community will continue to maintain this extension.

ParadoxV5, 2024-08-24:
> Hopefully someone from the community will continue to maintain this extension.

https://github.com/vscode-arduino/vscode-arduino

thelastoutpostworkshop, 2024-08-26:
I think we could still use VSCode without the extension by using the arduino CLI: https://github.com/arduino/arduino-cli
The VSCode Extension and the Arduino IDE are using it behind the scene to launch compilation, upload, etc.

davidlmorris, 2024-08-27:
Sounds like someone just accepted the challenge of integrating Arduino CLI into task runners et al for vscode, using arduino.json as a start and finding a way to incorporate all the libs et al in such a way that that c_cpp_properties.json, or something like, it is kept up to date.  Given the struggle that Platformio is having with the latest Esp32 Arduino libs (for Espressif 5.x), and MS giving up completely (though I bet that was "just" a [staffing issue](https://www.forbes.com/sites/emilsayegh/2024/08/19/the-great-tech-reset-unpacking-the-layoff-surge-of-2024/) ), I suspect there might be a little bit more to it.  

However, I would welcome a sound and bug-free result.  It is after all not an impossible feat, right?

wbadry, 2024-08-27:
Honestly , I would switch to [Visual Micro](https://www.visualmicro.com/) and Visual Studio since this extension sooner or later will be broken without dedicated maintenance.

The Arduino IDE is so awful in many ways compared to VSCode.

benreu, 2024-08-27:
> Honestly , I would switch to [Visual Micro](https://www.visualmicro.com/) and Visual Studio since this extension sooner or later will be broken without dedicated maintenance.

Are you affiliated with Visual Micro by any chance?

PeterWone, 2024-08-28:
> If you're not going to continue dev on it, then at least hand the project over to other developers. It can't be that a functioning extension that really makes sense simply stops.

Yes, it can. Microsoft did this to
- Windows Phone
- Silverlight
- MSN Messenger
- The entire Expression suite of image/web tools
- Movie Maker
- Windows Mail

The argument I once used against open source was that you can't rely on individuals to continue to maintain it. I have come to the realisation that I had it completely backward. Individuals can and often do pass the baton. Corporations don't do that and they are just as likely to kill projects. All it takes is a budget change, or a change of management.

# How to fix this

The project has a modified MIT license. All that is required to continue the project is for someone to pick up the torch and fork it. I can and will maintain this IF THERE IS ENOUGH SPONSORSHIP to abandon my day job. 

The question is not what Microsoft should do but what all of you should do: put your money where your mouth is. Even looking at how this might be coordinated will be a time sink, so let's start with a show of interest. Vote this up if you want the project continued enough to sponsor it for $10 a year. We need 5000 votes for it to go ahead. Leave some sort of contact in a reply.

ParadoxV5, 2024-08-28:
@PeterWone start here: https://github.com/vscode-arduino/vscode-arduino

PeterWone, 2024-08-29:
> @PeterWone start here: https://github.com/vscode-arduino/vscode-arduino

That's 36 commits behind.

A-J-Bauer, 2024-08-29:
Bummer, I used this together with the Microsoft Remote SSH extensions.
Having no other viable option I now switched to using the arduino-cli in the Terminal.
Given that settings up **c_cpp_properties.json** (not actually used for compiling) for a new board will be a hazzle, using the arduino-cli is not that bad really (Made a [gist](https://gist.github.com/A-J-Bauer/b20b775c6554b24c6283e3180b90d416) here with the steps required to set this up remotely on a Raspberry Pi for a Nano board).

benreu, 2024-08-29:
> against open source was that you can't rely on individuals to continue to maintain it. 

> Individuals can and often do pass the baton. Corporations don't do that

Therein is part of the answer. Individuals don't start something for market hype. They start a coding project because there is a need or a passion. That need or passion may go away, but they feel attached to the project/code.

chicodaman, 2024-08-29:
> Having no other viable option I now switched to using the arduino-cli in the Terminal.
> Given that settings up c_cpp_properties.json (not actually used for compiling) for a new board will be a [...]

Interesting. Thanks for writing the gist.

Would it be possible to show what a similar CLI sequence might look like for an ESP32 local compile?  For example, how to set up and compile an ESP32 Dev Module "blink" sketch in the local machine's Arduino project folder? 

Once that process is understood, perhaps some of the setup heavy-lifting might be handled by a scripting extension or something...

A-J-Bauer, 2024-08-30:
> Would it be possible to show what a similar CLI sequence might look like for an ESP32 local compile? For example, how to set up and compile an ESP32 Dev Module "blink" sketch in the local machine's Arduino project folder?

This [video](https://www.youtube.com/watch?v=J-qGn1eEidA&t=1s) shows how to create, compile and load a blink sketch.<br>The recommended ways of installing it: https://arduino.github.io/arduino-cli/1.0/installation/.

JohnKozell, 2024-08-30:
Just Microsoft being Microsoft.

juanker-cso, 2024-08-31:
Why?

MatthewSteeves, 2024-08-31:
> Why?

More explaination here: https://github.com/microsoft/vscode-arduino/issues/1760#issue-249789214

hanxiaoxin, 2024-09-01:
what a bad news. I love arduino extension,i don't now why does this decision

REDIDSOFT, 2024-09-02:
> Hi VS Code Arduino users, we make this announcement to inform you that we are deprecating the Arduino extension. We recommend existing customers to use the [Arduino IDE software](https://www.arduino.cc/en/software)

Shame on you Microsoft....Shame on you!

PeterWone, 2024-09-02:
@benreu I actually started my _own_ Arduino extension a few years back. I was fed up with the Microsoft one being borked every time VSCode and the extension got out of synch for their Node version, which prevents the native extensions from working thereby messing up the integration with the Arduino IDE. The solution was thunderingly obvious: shell the CLI instead. I said this to various people at Microsoft but it didn't seem to be happening, so I got a POC working and then life happened for a while, and then Microsoft actually _did_ it making mine a bit pointless.

But it wasn't all that hard to do. I notice there are only five upvotes on my request for expressions of support, so that's not likely to go anywhere. My interest is in using it as a custom controller for things like remote start and timing of diesel irrigation pumps that weren't designed for remote start. Projects like those reveal all the things that hobby users never encounter, like power management, filtering and battery level monitoring in a high temperature high vibration corrosive environment with IP67 enclosures and sealed connectors and how to distinguish engine vibrations from the slapping of waves on a pontoon. 

I had a look at your profile and it's nice to see someone else using them as more than a toy. If there isn't enough OSS support then we could look at sharing the maintenance burden. Either way, feel free to drop me a line at peter dot wone at outlook dot com.

lersi, 2024-09-03:
Hello all, [this fork](https://github.com/vscode-arduino/vscode-arduino) looks like the best candidate to continue maintenance of this extension.
In addition, this fork already has it's own presence on the market place: [https://marketplace.visualstudio.com/items?itemName=vscode-arduino.vscode-arduino-community](https://marketplace.visualstudio.com/items?itemName=vscode-arduino.vscode-arduino-community)

longware, 2024-09-10:
遗憾，[Arduino IDE software](https://www.arduino.cc/en/software) 就是兼容性好点，但是编辑器并不好用。

M0JXD, 2024-09-10:
I've nothing else to add that hasn't been mentioned, just wanted to comment to voice that I'm another upset user.
Thanks as always Microsoft. Embrace, extend and extinguish as usual.

avsavushkiny, 2024-09-15:
It is a pity that the extension is no longer supported. it was a very successful combination.

lersi, 2024-09-15:
> It is a pity that the extension is no longer supported. it was a very successful combination.

look at the bright side, the extension does not get any support from microsoft, but it does not mean that it is no longer supported at all, the community fork looks like it is still maintained.
Link to community fork:
https://github.com/vscode-arduino/vscode-arduino

jwalters1955, 2024-09-30:
Total pain in the proverbial. I'm working on a Linux platform and Arduino V2.XX is not available so stuck with 1.8. Have 2.3.2 on a PC but it's become so bloated it takes 10 minutes to compile anything.