microsoft/vscode-arduino #1760: VS Code Arduino Extension Removal from Marketplace

gcampbell-msft, 2024-08-30:
Dear Arduino Extensions customers,
 
Thank you for your continued support and enthusiasm for the Arduino extension and platform. Your feedback and engagement have been invaluable, and we’re incredibly proud of what we’ve built together.
 
As we looked toward the future, we wanted to ensure that we’re dedicating the necessary time and resources to provide the best possible experience for our users. After much deliberation, we’ve made the difficult decision to deprecate and remove the Arduino extension from the VSCode Marketplace.
 
This decision was not taken lightly, and it certainly wasn’t due to a lack of love for the project or the amazing community that has grown around it. On the contrary, we had ambitious plans for its future. However, after carefully evaluating our current priorities and capabilities, we realized that we wouldn’t be able to give Arduino the attention and updates it truly deserves.

This extension will no longer be available on the VS Code Marketplace as of 10/1/2024. We will also archive and no longer maintain this repository on that date. 

Please see the related deprecation details in this issue: #1757.

DRSDavidSoft, 2024-09-02:
This is a nightmare, I hope the community edition of the extension survives this hassle. Reminds me of the entire Arduino IDE fork that happened several years ago.

Anyways, Microsoft, thank you for creating and maintaining this amazing tool for so many years! ❤️ I hope it continues to thrive by other community members

https://github.com/vscode-arduino/vscode-arduino

lersi, 2024-09-03:
Hello all, [this fork](https://github.com/vscode-arduino/vscode-arduino) looks like the best candidate to continue maintenance of this extension.
In addition, this fork already has it's own presence on the market place: [https://marketplace.visualstudio.com/items?itemName=vscode-arduino.vscode-arduino-community](https://marketplace.visualstudio.com/items?itemName=vscode-arduino.vscode-arduino-community)

Benik3, 2024-09-04:
I was afraid of this :/
The Arduino IDE is still not close to the Visual Studio Code :/

JPZV, 2024-09-05:
> This extension will no longer be available on the VS Code Marketplace as of 10/1/2024

Use the `YYYY/MM/DD` date format god damn it.

Also, the Arduino IDE is not that good at all, neither the original (1.X) nor the new one (2.X). It's extremely so slow, bad optimized, ugly, and not having any feature that with VSCode we can have.

I hope the community extension will be a good replacement because I totally loved this extension and never used the Arduino IDE again (except for using exclusives addons for the 1.X)

brian3-21, 2024-09-10:
Could someone recommend another text editor that is widely used currently? I am already very familiar with VSCode, and from what I've tried, nothing compares to the Arduino IDE.

ParadoxV5, 2024-09-10:
@brian3-21 I recommend VSCode with https://marketplace.visualstudio.com/items?itemName=vscode-arduino.vscode-arduino-community

> [(comment)](#issuecomment-2324897987)

davidlmorris, 2024-09-11:
If you have more experience with coding and would like more control over your options while using (a more) standard C++, and don't mind being limited to out-of-date libraries (thinking Esp32 here), then Vscode with PlatformIO is ideal.

Otherwise, as expressed here numerous times, and unless the '[vscode-arduino-community](https://github.com/vscode-arduino/vscode-arduino)' fork takes off and remains safe, you are what is technically known as... ...screwed.

As it turns out someone is working on it, as there seem to have been updates in the last 24 hours.  So this may well become a viable option.

pigsy1983, 2024-09-18:
好伤心啊，i wish u can come back

rbeesley, 2024-10-01:
@davidlmorris, agree fully. I've been using PlatformIO, moving away from ArduinoIDE, and it works really well across I whole slew of platforms and boards. I had been watching this project as an alternative and I'm disappointed the Microsoft repo is being archived, but PlatformIO is very capable.